huggingface/lerobot #1538: issue training model

jeffrey-c-star, 2025-07-18:
Hello! When I try to train my model, it is giving me this error. 

"raise ValueError(f"Feature type '{_type}' not found. Available feature types: {list(_FEATURE_TYPES.keys())}")
ValueError: Feature type 'List' not found. Available feature types: ['Value', 'ClassLabel', 'Translation', 'TranslationVariableLanguages', 'LargeList', 'Sequence', 'Array2D', 'Array3D', 'Array4D', 'Array5D', 'Audio', 'Image', 'Video', 'Pdf', 'VideoFrame']
"

This is weird because it is able to train using one of my datasets but not the other. Could anyone help me out? Thanks!

michel-aractingi, 2025-07-18:
Can you provide more context, what are you running and with which dataset

genemerewether, 2025-07-19:
At least for me, this is the `datasets` python package not being limited to `<4.0`. After adding that version restriction, you also need to purge datasets downloaded when `datasets` had jumped to `4.0`

alexander-cobot, 2025-07-19:
@genemerewether you were right! Many thanks.

Liu-Xiaoyan97, 2025-07-22:
@alexander-cobot If I used the 4.0.0 version of datasets for data collection, how can I use this data for training in the 3.6 version environment? I found that even when upgrading the version to 4.0.0, errors still occur. The more concentrated errors are in torch.stack().
`TypeError: stack(): argument 'tensors' (position 1) must be tuple of Tensors, not Column`

Liu-Xiaoyan97, 2025-07-22:
> At least for me, this is the `datasets` python package not being limited to `<4.0`. After adding that version restriction, you also need to purge datasets downloaded when `datasets` had jumped to `4.0`

If I used the 4.0.0 version of datasets for data collection, how can I use this data for training in the 3.6 version environment? I found that even when upgrading the version to 4.0.0, errors still occur. The more concentrated errors are in torch.stack().
`TypeError: stack(): argument 'tensors' (position 1) must be tuple of Tensors, not Column`

Liu-Xiaoyan97, 2025-07-22:
I think I know the answer. I downloaded a dataset collected with a version no higher than 3.6.0 from hf and applied the data types in parquet to the dataset collected with version 4.0.0, which solved my problem. 
`import os
import pyarrow.parquet as pq

recomment_file_path = "lerobot/svla_so101_pickplace/data/chunk-000/file-000.parquet"
table2 = pq.read_table(recomment_file_path)
target_schema = table2.schema

dir = "your_own_dataset/data/chunk-000/"
target_dir = "your_own_dataset/data/chunk-001/"

os.makedirs(target_dir, exist_ok=True)

for filename in os.listdir(dir):
    if filename.endswith('.parquet'):
        file_path = os.path.join(dir, filename)
        try:
            table1 = pq.read_table(file_path)
            table1 = table1.cast(target_schema)
            target_file_path = os.path.join(target_dir, filename)
            pq.write_table(table1, target_file_path)
        except Exception as e:
            print(f"rasie an error when processing {file_path}: {e}")
`
delete the chunk-000 and then renamed chunk-001 to chunk-000.

jeffrey-c-star, 2025-07-23:
Hello! I tried some of the suggestions and am now getting this error 

 File "/mnt/shared-scratch/Shakkottai_S/jeffrey.chuang/miniconda3/envs/lerobot/lib/python3.10/dataclasses.py", line 1198, in fields
    raise TypeError('must be called with a dataclass type or instance') from None
TypeError: must be called with a dataclass type or instance

has anyone ever encountered this? Thanks!

wxkly8888, 2025-07-30:
I got this error too @jeffrey-c-star  have you fixed it?

jeffrey-c-star, 2025-07-30:
Howdy! I am using mobaxterm to train the model in a remote gpu, and I found out that the environment that I collected the dataset with and the mobaxterm were different. When I updated the version, it fixed the problem! I currently have an issue with deploying a policy and would really appreciate it if anyone could take a look at that issue. Thanks!

Tomorrowdawn, 2025-08-10:
I also ran into this problem, but with datasets==3.6.0. When I downgraded to 3.0.0, I encountered the same issue mentioned in [this comment](https://github.com/huggingface/lerobot/issues/1538#issuecomment-3108937083).

Interestingly, upgrading to 4.0.0 made everything work seamlessly.

It seems the core issue might be a mismatch between how the datasets are being generated and how they are being parsed. This is just a wild guess since I don't have much knowledge of these benchmarks.