huggingface/lerobot #1557: async inference "self._obs_sanity_checks(obs, self.last_processed_obs)"will nerver be True

Ukua, 2025-07-21:
When runing async inference with my own robot and run client and server in the same machine,using ACT policy
I find that whatever `chunk_size_threshold` i changed nerver worked,the server will always wait for `action_chunk` empty
So i find the check below in file `policy_server.py` will alway False until obs.must_go be True
```python
if (
    obs.must_go
    or self.last_processed_obs is None
    or self._obs_sanity_checks(obs, self.last_processed_obs)
):
```
Bypass it will fix this problem for me,maybe the method used in `_obs_sanity_checks` is not usable?

fracapuano, 2025-07-21:
Hey @Ukua 👋 thank you so much for the issue ⭐ May I ask you if you can provide a bit more details? We benchmarked async inference fairly thoroughly and don't have issues on our end, but happy to help debugging what's going on your end 🤗

closed 2025-07-22 by Ukua

LorenzoFerriniCodes, 2026-03-20:
Thank you for the quick answer @fracapuano! At PAL Robotics, we are deploying VLAs using LeRobot on our TIAGo Pro. We can see that, both in simulation and on the real robot, the async inference server behaves as above; it only allows observations to be processed when the explicit `must_go` related to the empty action queue is set, despite the client correctly  forwarding observations whenever the action queue length gets below the threshold. Unfortunately, it's hard at the moment to provide material to test our setting: we are working toward providing the tools to test and deploy VLAs on our robots, but it's still WiP. However, we checked the distance between the consecutive processed observations, and it never gets bigger than 1.0.

In general, I would see adding an explicit way to customize the parameter for the server at runtime quite useful. Depending on the specific embodiment, way smaller joint angle deltas might generate the same cartesian delta (this might actually be what we are seeing with our robot). Moreover, I think the targeted task defines the requirement for recomputation over smaller or bigger joint differences: for instance, higher precision tasks might require to perform inference over smaller joint angle variations, therefore requiring to decrease the value of the observation similarity threshold. I also think that users of the server might want to understand what happens when disabling the parameter (setting it negative).