huggingface/lerobot #1575: docker documentation needed

LokiMetaSmith, 2025-07-22:
### System Info

```Shell
The docker documentation is inadequate. 
There's a folder and no documentation on how to run it. 

this info should be in the main readme about how to get started with docker

# This Dockerfile is designed for a lerobot user who wants to
# experiment with the project. It starts from an Python Slim base image.

run in the root lerobot directory

# docker build -f docker/Dockerfile.user -t lerobot-user .
# docker run -it --rm lerobot-user
```

### Information

- [ ] One of the scripts in the examples/ folder of LeRobot
- [ ] My own task or dataset (give details below)

### Reproduction

there is no readme in the docker folder or reference to docker in the docs

### Expected behavior

there is enough information to either read other documentation sources for installing and using docker or the documentation is complete enough to run the included docker build scripts

heuzef, 2025-07-30:
Hi @LokiMetaSmith :) I search somes docs too. Did we talk about this one ? https://hub.docker.com/r/huggingface/lerobot-cpu

heuzef, 2025-07-31:
I have confirmation there is "no link" beetween the dockers files and hub image.

I try to add somes instruction: https://github.com/huggingface/lerobot/pull/1633

But i confirm, better to add a README file here.

CarolinePascal, 2026-01-09:
Hi @lakhmanisahil !

I'm not sure I've seen anyone contributing further than what @heuzef did on LeRobot support in Docker... So I guess this is still an open issue !

Beware, it is tougher than it seems,

Best,

Caroline.