huggingface/lerobot #1578: Lerobot metaworld dataset only provides 49 tasks

chenkang455, 2025-07-23:
https://huggingface.co/datasets/lerobot/metaworld_mt50

There are only 49 tasks and "Push the puck to a goal" task repeates twice

chenkang455, 2025-07-23:
> I also noticed similar issues. Do you have any recommended vla evaluation codebases for metaworld that I can refer to?

I am trying pi0 on it, is that suitable for you?

Zooy138, 2025-07-27:
> https://huggingface.co/datasets/lerobot/metaworld_mt50
> 
> There are only 49 tasks and "Push the puck to a goal" task repeates twice

I also noticed this issue. Could you clarify how π₀ was trained and evaluated on MT50 given the incomplete dataset missing one task's data.

chenkang455, 2025-07-28:
> > https://huggingface.co/datasets/lerobot/metaworld_mt50
> > 
> > There are only 49 tasks and "Push the puck to a goal" task repeates twice
> 
> I also noticed this issue. Could you clarify how π₀ was trained and evaluated on MT50 given the incomplete dataset missing one task's data.

This dataset doesn't miss one task, it has 50 tasks but two tasks are the same...

closed 2026-02-24 by github-actions[bot]