huggingface/lerobot #1580: Environment_State in act and SmolVLA policy

kasiv008, 2025-07-24:
Hi, Thanks for the awesome work!
I have been noticing a variable called observation.environment_state in the act policy. What is exactly the feature environment_state. Thanks!

atyshka, 2025-07-24:
I think it's intended for sim environments to directly leverage object positions rather than using perception from pixels

kasiv008, 2025-07-24:
Okay that's cool. I was looking towards a task pointer while training ACTpolicy on a dataset with different task and input the specific task/task_index while inference. This way I don't have to train a different policy for each task.

jadechoghari, 2025-10-08:
Closing this issue as it has been addressed
Feel free to open a new one if you run into anything else!

closed 2025-10-08 by jadechoghari