huggingface/lerobot #1585: PermissionError: [Errno 13] on Windows when using lerobot record

krayF1sh, 2025-07-24:
Hello,

I am getting a PermissionError: [Errno 13] Permission denied when trying to run a command like lerobot record. The error occurs when the script tries to access a temporary file.

**Original Error Traceback:**
```
WARNING:root:No accelerated backend detected. Using default cpu, this will be slow.
WARNING:root:Device 'cuda' is not available. Switching to 'cpu'.
Traceback (most recent call last):
  File "C:\Users\vagge\miniconda3\envs\lerobot\lib\runpy.py", line 196, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "C:\Users\vagge\miniconda3\envs\lerobot\lib\runpy.py", line 86, in _run_code
    exec(code, run_globals)
  File "C:\Users\vagge\Documents\00_Projects\lerobot\src\lerobot\record.py", line 367, in <module>
    record()
  File "C:\Users\vagge\Documents\00_Projects\lerobot\src\lerobot\configs\parser.py", line 225, in wrapper_inner
    cfg = draccus.parse(config_class=argtype, config_path=config_path, args=cli_args)
  File "C:\Users\vagge\miniconda3\envs\lerobot\lib\site-packages\draccus\argparsing.py", line 211, in parse
    return parser.parse_args(args)
  File "C:\Users\vagge\miniconda3\envs\lerobot\lib\site-packages\draccus\argparsing.py", line 102, in parse_args
    args, _ = self.parse_known_args(args, namespace, is_parse_args=True)
  File "C:\Users\vagge\miniconda3\envs\lerobot\lib\site-packages\draccus\argparsing.py", line 136, in parse_known_args
    parsed_t = self._postprocessing(parsed_args)
  File "C:\Users\vagge\miniconda3\envs\lerobot\lib\site-packages\draccus\argparsing.py", line 180, in _postprocessing
    cfg = decoding.decode(self.config_class, deflat_d)
  File "C:\Users\vagge\miniconda3\envs\lerobot\lib\site-packages\draccus\parsers\registry_utils.py", line 78, in wrapper
    return base_func(*args, **kw)
  File "C:\Users\vagge\miniconda3\envs\lerobot\lib\site-packages\draccus\parsers\decoding.py", line 48, in decode
    return get_decoding_fn(cls)(raw_value, ())  # type: ignore
  File "C:\Users\vagge\miniconda3\envs\lerobot\lib\site-packages\draccus\parsers\decoding.py", line 159, in decode_dataclass
    instance = origin(**init_args)
  File "<string>", line 10, in __init__
  File "C:\Users\vagge\Documents\00_Projects\lerobot\src\lerobot\record.py", line 151, in __post_init__
    self.policy = PreTrainedConfig.from_pretrained(policy_path, cli_overrides=cli_overrides)
  File "C:\Users\vagge\Documents\00_Projects\lerobot\src\lerobot\configs\policies.py", line 206, in from_pretrained
    return draccus.parse(orig_config.__class__, config_file, args=cli_overrides)
  File "C:\Users\vagge\miniconda3\envs\lerobot\lib\site-packages\draccus\argparsing.py", line 211, in parse
    return parser.parse_args(args)
  File "C:\Users\vagge\miniconda3\envs\lerobot\lib\site-packages\draccus\argparsing.py", line 102, in parse_args
    args, _ = self.parse_known_args(args, namespace, is_parse_args=True)
  File "C:\Users\vagge\miniconda3\envs\lerobot\lib\site-packages\draccus\argparsing.py", line 136, in parse_known_args
    parsed_t = self._postprocessing(parsed_args)
  File "C:\Users\vagge\miniconda3\envs\lerobot\lib\site-packages\draccus\argparsing.py", line 173, in _postprocessing
    with open(config_path, "r") as f:
PermissionError: [Errno 13] Permission denied: 'C:\\Users\\vagge\\AppData\\Local\\Temp\\tmpc6eiq33f'
```
**Troubleshooting Steps Taken:**

I have confirmed this is not a system configuration issue.

The error occurred with both my original antivirus (Norton) and with the default Windows Defender (after a full Norton removal and system reboot).

Correct exclusions for the project folder and the Python executable were added to Windows Defender, but the error persisted.

The error occurs in both the default AppData\Local\Temp folder and a brand new custom folder (C:\lerobot-temp).

Windows "Controlled folder access" is turned off.

My system is running on Windows 11. Please let me know if you need any more information..

IsaacSinn, 2025-08-02:
Same issue here

IsaacSinn, 2025-08-03:
Thanks it worked

CarolinePascal, 2025-08-03:
Hi @Sachin-2007,

Thank you so much for the fix ! I believe you just saved all our Windows user ;) Would you be interested in opening a Pull Request with the proposed fix ? 👀 

Best,

Caroline.

aisy31, 2025-10-02:
I tried using the fix you gave but I seem to encounter the same problem again