huggingface/lerobot #1594: Error when loading dataset with dtype="image" features

destroy314, 2025-07-25:
### System Info

```Shell
lerobot at 862a4439ea4ce4671b84b713f28e705d0b6e172b

- `lerobot` version: 0.1.0
- Platform: Linux-5.4.0-212-generic-x86_64-with-glibc2.35
- Python version: 3.10.13
- Huggingface_hub version: 0.33.4
- Dataset version: 2.19.0
- Numpy version: 1.26.3
- PyTorch version (GPU?): 2.7.1+cu126 (True)
- Cuda version: 12060
- Using GPU in script?: False
```

### Information

- [ ] One of the scripts in the examples/ folder of LeRobot
- [x] My own task or dataset (give details below)

### Reproduction

When using LeRobotDataset(repo_id="", root=...) to load episodes that contain features with dtype="image", I encountered a runtime error during data access. It appears that image fields are not properly decoded before being passed to hf_transform_to_torch, leading to a RuntimeError.

```
lerobot_dataset = LeRobotDataset('','path/to/test_task')
lerobot_dataset[0]
```

```
  File "/opt/conda/lib/python3.10/site-packages/lerobot/datasets/lerobot_dataset.py", line 712, in __getitem__
    item = self.hf_dataset[idx]
  File "/opt/conda/lib/python3.10/site-packages/datasets/arrow_dataset.py", line 2861, in __getitem__
    return self._getitem(key)
  File "/opt/conda/lib/python3.10/site-packages/datasets/arrow_dataset.py", line 2846, in _getitem
    formatted_output = format_table(
  File "/opt/conda/lib/python3.10/site-packages/datasets/formatting/formatting.py", line 633, in format_table
    return formatter(pa_table, query_type=query_type)
  File "/opt/conda/lib/python3.10/site-packages/datasets/formatting/formatting.py", line 397, in __call__
    return self.format_row(pa_table)
  File "/opt/conda/lib/python3.10/site-packages/datasets/formatting/formatting.py", line 486, in format_row
    formatted_batch = self.format_batch(pa_table)
  File "/opt/conda/lib/python3.10/site-packages/datasets/formatting/formatting.py", line 516, in format_batch
    return self.transform(batch)
  File "/opt/conda/lib/python3.10/site-packages/lerobot/datasets/utils.py", line 279, in hf_transform_to_torch
    items_dict[key] = [x if isinstance(x, str) else torch.tensor(x) for x in items_dict[key]]
  File "/opt/conda/lib/python3.10/site-packages/lerobot/datasets/utils.py", line 279, in <listcomp>
    items_dict[key] = [x if isinstance(x, str) else torch.tensor(x) for x in items_dict[key]]
RuntimeError: Could not infer dtype of dict
```

The image feature appears as a dictionary with `{"bytes": ..., "path": ...}` when passed to the transform. A sample dataset can be found at [here](https://drive.google.com/file/d/14yzm3eJIOnw_eMG4P3KEExL6gXAOt8bw/view?usp=share_link). This might be a configuration issue on my side, as the same operation works fine in another environment. Apologies in advance!

### Expected behavior

No exceptions are generated

lichenlong2024, 2025-11-02:
have you fix it ,I also this question

imstevenpmwork, 2026-03-11:
Hello, 

The version of `lerobot` used in the report of this ticket seems quite out-dated, so I'm closing this as I believe this is not longer an issue in the latest `main`. However, feel free to reopen if this is still happening.

Thanks!

closed 2026-03-11 by imstevenpmwork