huggingface/lerobot #1620: Policy image features name mismatch

RiccardoIzzo, 2025-07-29:
### System Info

```Shell
- `lerobot` version: 0.2.0
- Platform: Linux-6.8.0-64-generic-x86_64-with-glibc2.35
- Python version: 3.10.13
- Huggingface_hub version: 0.34.0
- Dataset version: 3.6.0
- Numpy version: 1.26.4
- PyTorch version (GPU?): 2.7.1+cu126 (True)
- Cuda version: 12060
- Using GPU in script?: NVIDIA GeForce RTX 3070 Laptop
```

### Information

- [ ] One of the scripts in the examples/ folder of LeRobot
- [x] My own task or dataset (give details below)

### Reproduction

From the [async inference example](https://huggingface.co/docs/lerobot/en/async?start_robot_client=API+example):
```
import threading
from lerobot.robots.so101_follower import SO101FollowerConfig
from lerobot.cameras.opencv.configuration_opencv import OpenCVCameraConfig
from lerobot.scripts.server.configs import RobotClientConfig
from lerobot.scripts.server.robot_client import RobotClient
from lerobot.scripts.server.helpers import visualize_action_queue_size

# 1. Create the robot instance
"""Check out the cameras available in your setup by running `python lerobot/find_cameras.py`"""
# these cameras must match the ones expected by the policy
# check the config.json on the Hub for the policy you are using
camera_cfg = {
    "image2": OpenCVCameraConfig(index_or_path="/dev/video2", width=640, height=480, fps=30),
    "image": OpenCVCameraConfig(index_or_path="/dev/video1", width=640, height=480, fps=30),
    "image3": OpenCVCameraConfig(index_or_path="/dev/video3", width=640, height=480, fps=30),
}

robot_cfg = SO101FollowerConfig(
  port="/dev/ttyACM0",
  id="my_follower_arm",
  cameras=camera_cfg
)

# 3. Create client configuration
client_cfg = RobotClientConfig(
    robot=robot_cfg,
    server_address="localhost:8080",
    policy_device="cuda",
    policy_type="smolvla",
    # pretrained_name_or_path="helper2424/smolvla_check_async",
    pretrained_name_or_path="lerobot/smolvla_base",
    chunk_size_threshold=0.5,
    actions_per_chunk=50,  # make sure this is less than the max actions of the policy
)

# 4. Create and start client
client = RobotClient(client_cfg)

# 5. Specify the task
task = "Don't do nothing, stay still"

if client.start():
    # Start action receiver thread
    action_receiver_thread = threading.Thread(target=client.receive_actions, daemon=True)
    action_receiver_thread.start()

    try:
        # Run the control loop
        client.control_loop(task)
    except KeyboardInterrupt:
        client.stop()
        action_receiver_thread.join()
        # (Optionally) plot the action queue size
        visualize_action_queue_size(client.action_queue_size)
```

The same problem also occurs from the command line:
```
python -m lerobot.scripts.server.robot_client \
    --server_address=127.0.0.1:8080 \
    --robot.type=so101_follower \
    --robot.port=/dev/ttyACM0 \
    --robot.id=my_follower_arm \
    --robot.cameras="{
        image2: {type: opencv, index_or_path: /dev/video2, width: 640, height: 480, fps: 30}, \
        image:  {type: opencv, index_or_path: /dev/video1, width: 640, height: 480, fps: 30}, \
        image3: {type: opencv, index_or_path: /dev/video3, width: 640, height: 480, fps: 30}  \
        }" \
    --task="Don't do anything, stay still" \
    --policy_type=smolvla \
    --pretrained_name_or_path=lerobot/smolvla_base \
    --policy_device=mps \
    --actions_per_chunk=50 \
    --chunk_size_threshold=0.5 \
    --aggregate_fn_name=weighted_average \
    --debug_visualize_queue_size=True
```

### Expected behavior

The code appends a "-images" prefix in the dictionary, resulting in an` AssertionError: Policy image features must match robot cameras! Received ['observation.image2', 'observation.image', 'observation.image3'] != ['observation.images.image2', 'observation.images.image', 'observation.images.image3']`

```
Traceback (most recent call last):
  File "/home/riccardo/anaconda3/envs/lerobot/lib/python3.10/runpy.py", line 196, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/home/riccardo/anaconda3/envs/lerobot/lib/python3.10/runpy.py", line 86, in _run_code
    exec(code, run_globals)
  File "/home/riccardo/lerobot/src/lerobot/scripts/server/robot_client.py", line 512, in <module>
    async_client()  # run the client
  File "/home/riccardo/anaconda3/envs/lerobot/lib/python3.10/site-packages/draccus/argparsing.py", line 225, in wrapper_inner
    response = fn(cfg, *args, **kwargs)
  File "/home/riccardo/lerobot/src/lerobot/scripts/server/robot_client.py", line 488, in async_client
    client = RobotClient(cfg)
  File "/home/riccardo/lerobot/src/lerobot/scripts/server/robot_client.py", line 105, in __init__
    validate_robot_cameras_for_policy(lerobot_features, policy_image_features)
  File "/home/riccardo/lerobot/src/lerobot/scripts/server/helpers.py", line 63, in validate_robot_cameras_for_policy
    assert set(image_keys) == set(policy_image_features.keys()), (
AssertionError: Policy image features must match robot cameras! Received ['observation.image2', 'observation.image', 'observation.image3'] != ['observation.images.image2', 'observation.images.image', 'observation.images.image3']
```

The provided dict should match the image feature names of the policies, however for this reason there is a mismatch.
This problem occurs independently of the number of cameras given.

RiccardoIzzo, 2025-07-29:
Already tried the fix suggested in the LeRobot Discord of removing `images` in `features[f"{prefix}.images.{key}"]` from `def hw_to_dataset_features()` in `lerobot/datasets/utils.py`. 
However, this causes a side effect as the system is not able to find the cameras anymore.
`AssertionError: Policy image features must match robot cameras! Received ['observation.image2', 'observation.image', 'observation.image3'] != [])`

sidlak-c137, 2025-07-30:
Getting the same issue, also tried setting `OBS_IMAGES = "observation"` in the constants, and that got further, but the server gets `Error in StreamActions`

ohharsen, 2025-08-05:
I was able to make some progress with
```
  --robot.cameras='{ image: {type: opencv, index_or_path: 0, width: 1280, height: 720, fps: 30},
                     image2:{type: opencv, index_or_path: 0, width: 1280, height: 720, fps: 30},
                     image3:{type: opencv, index_or_path: 0, width: 1280, height: 720, fps: 30} }'
```

But also getting `Error in StreamActions`

AdilZouitine, 2025-08-05:
Hey, we’re aware of this issue. With the Pipeline PR—which will be merged this week or next—you’ll be able to rename image features on the fly.
Inference with pretrained VLAs like smolVLA and PI0 will also be seamless.
Thanks for your patience! 😄

PRs, https://github.com/huggingface/lerobot/pull/1431 https://github.com/huggingface/lerobot/pull/1452 https://github.com/huggingface/lerobot/pull/1645 https://github.com/huggingface/lerobot/pull/1659

jadechoghari, 2025-10-17:
As Adil mentioned, Pipeline fixed this issue! 
Follow the documentation to rename your observation (the preprocess step) and feel free to let us know (open the issue again) if you encounter any blocker!
https://huggingface.co/docs/lerobot/en/introduction_processors

closed 2025-10-17 by jadechoghari