huggingface/lerobot #1624: PermissionError: Cannot create temporary files on Windows (even after setting TEMP/TMP)

YouXiangyu, 2025-07-30:
### Problem Description

I am encountering a `PermissionError` when running the `lerobot/scripts/eval.py` script on Windows. The error occurs when the script (or an underlying library like `draccus`) attempts to create temporary files.

Crucially, this error persists even after I have explicitly set the `TEMP` and `TMP` environment variables to a directory on my D: drive (`D:\Temp`), where I have confirmed sufficient disk space. The error message indicates a permission denial for this newly designated temporary directory.

### Steps to Reproduce

1.  **Operating System:** Windows [Your Windows Version, e.g., 10 Pro, 11 Home]
2.  **Python Version:** [Your Python Version, e.g., 3.10.12]
3.  **`lerobot` Installation:** [How you installed lerobot, e.g., "Cloned the `main` branch and installed with `pip install -e .`"]
4.  **Environment Setup:**
    *   Created a directory `D:\Temp`.
    *   Set system/user environment variables (or temporarily in cmd/PowerShell) as follows:
        ```
        set TEMP=D:\Temp
        set TMP=D:\Temp
        ```
        (or `$env:TEMP="D:\Temp"; $env:TMP="D:\Temp"` for PowerShell)
    *   Ensured my user account has Full Control permissions on `D:\Temp` via folder properties -> Security tab.
5.  **Command Executed:**
    ```bash
    python -m lerobot.scripts.eval \
        --policy.path=lerobot/diffusion_pusht \
        --env.type=pusht \
        --eval.batch_size=10 \
        --eval.n_episodes=10 \
        --policy.use_amp=false \
        --policy.device=cuda
    ```

### Expected Behavior

The `lerobot.scripts.eval` script should execute successfully, creating any necessary temporary files in the `D:\Temp` directory without encountering a `PermissionError`.

### Actual Behavior

The script terminates with a `PermissionError`, indicating it cannot write to the `D:\Temp` directory.

### Error Traceback
Traceback (most recent call last):
File "<frozen runpy>", line 198, in _run_module_as_main
File "<frozen runpy>", line 88, in _run_code
File "D:\tools\lerobot\lerobot\src\lerobot\scripts\eval.py", line 506, in <module>
eval_main()
File "D:\tools\lerobot\lerobot\src\lerobot\configs\parser.py", line 224, in wrapper_inner
cfg = draccus.parse(config_class=argtype, config_path=config_path, args=cli_args)
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "D:\tools\miniconda\envs\lerobot\Lib\site-packages\draccus\argparsing.py", line 211, in parse
return parser.parse_args(args)
^^^^^^^^^^^^^^^^^^^^^^^
File "D:\tools\miniconda\envs\lerobot\Lib\site-packages\draccus\argparsing.py", line 102, in parse_args
args, _ = self.parse_known_args(args, namespace, is_parse_args=True)
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "D:\tools\miniconda\envs\lerobot\Lib\site-packages\draccus\argparsing.py", line 136, in parse_known_args
parsed_t = self._postprocessing(parsed_args)
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "D:\tools\miniconda\envs\lerobot\Lib\site-packages\draccus\argparsing.py", line 180, in _postprocessing
cfg = decoding.decode(self.config_class, deflat_d)
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "D:\tools\miniconda\envs\lerobot\Lib\site-packages\draccus\parsers\registry_utils.py", line 78, in wrapper
return base_func(*args, **kw)
^^^^^^^^^^^^^^^^^^^^^^
File "D:\tools\miniconda\envs\lerobot\Lib\site-packages\draccus\parsers\decoding.py", line 48, in decode
return get_decoding_fn(cls)(raw_value, ()) # type: ignore
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "D:\tools\miniconda\envs\lerobot\Lib\site-packages\draccus\parsers\decoding.py", line 159, in decode_dataclass
instance = origin(**init_args)
^^^^^^^^^^^^^^^^^^^
File "<string>", line 9, in init
File "D:\tools\lerobot\lerobot\src\lerobot\configs\eval.py", line 43, in post_init
self.policy = PreTrainedConfig.from_pretrained(policy_path, cli_overrides=cli_overrides)
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "D:\tools\lerobot\lerobot\src\lerobot\configs\policies.py", line 206, in from_pretrained
return draccus.parse(orig_config.class, config_file, args=cli_overrides)
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "D:\tools\miniconda\envs\lerobot\Lib\site-packages\draccus\argparsing.py", line 211, in parse
return parser.parse_args(args)
^^^^^^^^^^^^^^^^^^^^^^^
File "D:\tools\miniconda\envs\lerobot\Lib\site-packages\draccus\argparsing.py", line 102, in parse_args
args, _ = self.parse_known_args(args, namespace, is_parse_args=True)
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "D:\tools\miniconda\envs\lerobot\Lib\site-packages\draccus\argparsing.py", line 136, in parse_known_args
parsed_t = self._postprocessing(parsed_args)
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "D:\tools\miniconda\envs\lerobot\Lib\site-packages\draccus\argparsing.py", line 173, in _postprocessing
with open(config_path, "r") as f:
^^^^^^^^^^^^^^^^^^^^^^
PermissionError: [Errno 13] Permission denied: 'D:\Temp\tmpenctwm7r'

### Additional Context

*   I initially encountered a similar `PermissionError` pointing to a C: drive temporary path, which led me to redirect `TEMP` and `TMP` to `D:\Temp`.
*   I have verified that there is ample free space on the D: drive.
*   I have explicitly granted "Full Control" permissions to my user account for the `D:\Temp` folder.
*   Running the command prompt as Administrator does not resolve the issue (though I prefer not to run as admin if possible).

Any guidance on how to resolve this persistent permission issue would be greatly appreciated.

Sachin-2007, 2025-08-02:
https://github.com/huggingface/lerobot/issues/1585#issuecomment-3146556355

IsaacSinn, 2025-08-03:
Thanks it worked