huggingface/lerobot #1626: Running ACT training error:     raise TypeError('must be called with a dataclass type or instance') from None TypeError: must be called with a dataclass type or instance

samuelp4065, 2025-07-30:
Hey, I tried the solution, I was still getting this error. 

Traceback (most recent call last):
  File "/home/sw/lerobot/lerobot/scripts/train.py", line 288, in <module>
    train()
  File "/home/sw/lerobot/lerobot/configs/parser.py", line 227, in wrapper_inner
    response = fn(cfg, *args, **kwargs)
  File "/home/sw/lerobot/lerobot/scripts/train.py", line 128, in train
    dataset = make_dataset(cfg)
  File "/home/sw/lerobot/lerobot/common/datasets/factory.py", line 90, in make_dataset
    dataset = LeRobotDataset(
  File "/home/sw/lerobot/lerobot/common/datasets/lerobot_dataset.py", line 499, in __init__
    self.hf_dataset = self.load_hf_dataset()
  File "/home/sw/lerobot/lerobot/common/datasets/lerobot_dataset.py", line 622, in load_hf_dataset
    hf_dataset = load_dataset("parquet", data_dir=path, split="train")
  File "/home/sw/miniconda3/envs/lerobot/lib/python3.10/site-packages/datasets/load.py", line 2096, in load_dataset
    builder_instance.download_and_prepare(
  File "/home/sw/miniconda3/envs/lerobot/lib/python3.10/site-packages/datasets/builder.py", line 924, in download_and_prepare
    self._download_and_prepare(
  File "/home/sw/miniconda3/envs/lerobot/lib/python3.10/site-packages/datasets/builder.py", line 977, in _download_and_prepare
    split_generators = self._split_generators(dl_manager, **split_generators_kwargs)
  File "/home/sw/miniconda3/envs/lerobot/lib/python3.10/site-packages/datasets/packaged_modules/parquet/parquet.py", line 58, in _split_generators
    self.info.features = datasets.Features.from_arrow_schema(pq.read_schema(f))
  File "/home/sw/miniconda3/envs/lerobot/lib/python3.10/site-packages/datasets/features/features.py", line 1791, in from_arrow_schema
    metadata_features = Features.from_dict(metadata["info"]["features"])
  File "/home/sw/miniconda3/envs/lerobot/lib/python3.10/site-packages/datasets/features/features.py", line 1829, in from_dict
    obj = generate_from_dict(dic)
  File "/home/sw/miniconda3/envs/lerobot/lib/python3.10/site-packages/datasets/features/features.py", line 1455, in generate_from_dict
    return {key: generate_from_dict(value) for key, value in obj.items()}
  File "/home/sw/miniconda3/envs/lerobot/lib/python3.10/site-packages/datasets/features/features.py", line 1455, in <dictcomp>
    return {key: generate_from_dict(value) for key, value in obj.items()}
  File "/home/sw/miniconda3/envs/lerobot/lib/python3.10/site-packages/datasets/features/features.py", line 1470, in generate_from_dict
    field_names = {f.name for f in fields(class_type)}
  File "/home/sw/miniconda3/envs/lerobot/lib/python3.10/dataclasses.py", line 1198, in fields
    raise TypeError('must be called with a dataclass type or instance') from None
TypeError: must be called with a dataclass type or instance

For more details. Following is the solution I tried.

1. `pip install datasets==3.0`
2. Go to **pyproject.toml**
3. Change the dependencies for datasets.
4. ""datasets>=2.19.0,<4.0.0

However I still had the same error

Did anyone solve this problem?

xk21520, 2025-08-06:
我也遇到了一模一样的问题，请问最后怎么解决的

[image]

xk21520, 2025-08-07:
Could you show me the exact commands for creating the new database?