huggingface/lerobot #1627: Diffusion policy training at recent commit may not working properly

HiroIshida, 2025-07-30:
Relobot on branch 67196c9d5344cd932612cef79229f9d04134c91e

I tried push_t example in examples directory. I trained own policy and test is using 
2_evaluate_pretrained_policy.py  3_train_policy.py, where the former is tweaked only the policy path to load the learned policy, and the latter is not modified at all except for the number of steps.

The result indicates that, with number of steps = 5000 (defualt), it fails completely

https://github.com/user-attachments/assets/aa6c7891-eb23-4cad-9844-0462cdb8747a

and even with steps = 30000, it also fails completely

https://github.com/user-attachments/assets/31e29e8e-6cd2-4f9f-83a9-1807ff34605d

Does anybody encounter similar issue? 

## context (have not completely confirmed though..)
Actually, the reason why I did this is that even simple reaching task (solve IK and straight-ilne interpolation to generate demo trajectory) with 300 episodes in pybullet environment, diffusion policy is not trained well. 
Although when I tried 1 years ago, it works with 100 epoch to some extent.

nepyope, 2025-08-13:
I ran: 

```bash
lerobot-train \
  --dataset.repo_id=lerobot/pusht \
  --policy.type=diffusion \
  --steps=5000 \
  --policy.device=cuda \
  --batch_size=1024 \
  --output_dir=runs/pusht_dp_5k_1024 \
  --policy.push_to_hub=false \
  --wandb.enable=true \
  --wandb.project=diffusion_test
```

Then for eval: 

```bash
python -m lerobot.scripts.eval \
  --policy.path=checkpoints/last/pretrained_model \
  --env.type=pusht \
  --eval.batch_size=4 \
  --eval.n_episodes=10 \
  --policy.device=cpu \
  --policy.use_amp=false \
  --output_dir=outputs/eval_last
```

Results:

[image]

[image]
[image]
[image]

nepyope, 2025-08-13:
hash commit: 0878c6880fa4fbadf0742751cf7b015f2d63a769

closed 2025-08-13 by nepyope

nepyope, 2025-08-13:
yeah seems like that's the only difference, it makes sense because the number of steps is still 5000 meaning that compared to batch 64 batch 1024 sees 16 times more data