huggingface/lerobot #1654: Release 0.4.0

imstevenpmwork, 2025-08-02:
A Github Issue created for the upcoming release to discuss the planned features & changes:

nahidalam, 2025-08-04:
Please bump `transformers` and related version to support newer models like `gemma3` etc.

michel-aractingi, 2025-08-05:
LeRobotDataset V3: https://github.com/huggingface/lerobot/pull/1412
LeRobotDatasetStreaming: https://github.com/huggingface/lerobot/pull/1613

jcoleharrison, 2025-08-09:
Would love for someone on the HF side to validate  #1534 teleop for both Koch leader and follower before merging.  @CarolinePascal has been working on this but there may still be motor inversion issues that she's helping resolve before merging :)

chabir, 2025-10-24:
would you be able to look at the gpu slowness [https://github.com/huggingface/lerobot/issues/2282] ? my experience with collab is 4+x slower with A100 between 0.3.4 and 0.3.3 on a simple act training.

imstevenpmwork, 2025-10-24:
@chabir That's indeed one of our urgent items in our todo list. Will take some investigation and time, but we will definitely address it !

imstevenpmwork, 2025-11-03:
Hi team,

A huge thank you to everyone who contributed their valuable ideas and feedback on this thread. Your input has been essential.

With the release of v0.4.0, we're now ready to close this issue. You can check out the new release here:
https://github.com/huggingface/lerobot/releases/tag/v0.4.0

To keep things organized, I'll be moving any remaining open items into new, dedicated tickets. I'll also create a new issue to start tracking our progress for the next release.

Thanks again for your great work!