huggingface/lerobot #1675: TimeoutError: Timed out waiting for frame from camera OpenCVCamera(0) after 200 ms. Read thread alive: True

CrossyNautilus, 2025-08-06:
### System Info

```Shell
LeRobot version: f771e3eaf1b30aae65768513b2e67d351f5f6c2f
OS: MacOS Ventura 13.3.1
Python version: 3.10.18
Numpy version: 2.2.6
Torch version: 2.7.1
LeRobot configuration:
`lerobot` version: 0.3.2
- Platform: macOS-13.3.1-arm64-arm-64bit
- Python version: 3.10.18
- Huggingface_hub version: 0.34.3
- Dataset version: 3.6.0
- Numpy version: 2.2.6
- PyTorch version (GPU?): 2.7.1 (False)
- Cuda version: N/A
- Using GPU in script?: <fill in>
```

### Information

- [x] One of the scripts in the examples/ folder of LeRobot
- [ ] My own task or dataset (give details below)

### Reproduction

python -m lerobot.record \
    --robot.type=so100_follower \
    --robot.port=/dev/tty.usbmodem58CD1770331 \
    --robot.id=my_awesome_follower_arm \
    --robot.cameras="{ wrist.right: {type: opencv, index_or_path: 0, width: 1920, height: 1080, fps: 30}, top: {type: opencv, index_or_path: 1, width: 1920, height: 1080, fps: 30}}" \
    --teleop.type=so100_leader \
    --teleop.port=/dev/tty.usbmodem58CD1770511 \
    --teleop.id=my_awesome_leader_arm \
    --display_data=true \
    --dataset.repo_id=${HF_USER}/record-test \
    --dataset.num_episodes=5 \
    --dataset.single_task="Stack the yellow birdie"

### Expected behavior

I expect to be able to record a dataset, but it randomly disconnects between 5-20 seconds after running the script above and displays the error message.

forgetwhatuwant, 2025-08-06:
---
## **Step 1: Check Your Camera Formats**
```bash
v4l2-ctl --device=/dev/video0 --list-formats-ext
v4l2-ctl --device=/dev/video1 --list-formats-ext
```

Look for the output - this is what we need.

## **Step 2: Use MJPEG Format in Your Command**
check this pr https://github.com/huggingface/lerobot/pull/1558

```bash
python -m lerobot.record \
--robot.type=so100_follower \
--robot.port=/dev/tty.usbmodem58CD1770331 \
--robot.id=my_awesome_follower_arm \
--robot.cameras="{ wrist.right: {type: opencv, index_or_path: 0, width: 1920, height: 1080, fps: 30, fourcc: 'MJPG'}, top: {type: opencv, index_or_path: 1, width: 1920, height: 1080, fps: 30, fourcc: 'MJPG'}}" \
--teleop.type=so100_leader \
--teleop.port=/dev/tty.usbmodem58CD1770511 \
--teleop.id=my_awesome_leader_arm \
--display_data=true \
--dataset.repo_id=${HF_USER}/record-test \
--dataset.num_episodes=5 \
--dataset.single_task="Stack the yellow birdie"
```

**The key change:** Added `fourcc: 'MJPG'` to both cameras.

## **Why This Fixes It:**
- Your cameras support MJPEG (check with v4l2-ctl)
- MJPEG is much more stable for high resolution

That's it! Try this and let me know if it works.

CrossyNautilus, 2025-08-11:
Thanks for the reply!

>v4l2-ctl --device=/dev/video0 --list-formats-ext

I don't have v4l2-ctl installed on my Mac, how can I work around this?

When I run your lerobot.record command, I get the following error:
>draccus.utils.DecodingError: `robot.cameras.wrist.right`: The fields `fourcc` are not valid for OpenCVCameraConfig

Let me know what else I can try!

forgetwhatuwant, 2025-08-11:
---
## **Step 1: u can try this**
```bash
lerobot-find-cameras opencv
```

Look for the output, check ur camera supporting format.

## **Step 2: pls check pr first, that's a demo command**
check this pr https://github.com/huggingface/lerobot/pull/1558

**The key change:** Added `fourcc: 'MJPG'` to both cameras.

That's it! Try this and let me know if it works.

CrossyNautilus, 2025-08-31:
Thanks for your help, I was able to get the episode to record without timeout errors. When do you plan to merge the PR?

forgetwhatuwant, 2025-09-01:
good for u!

at1047, 2025-10-29:
@jzchen2 u can hardcode the fourcc option in the code, since the python file doesn't take it as input, but you need it for not to timeout

jzchen2, 2025-10-29:
@at1047 Would you happen to know which python file it is exactly? I tried looking for the file and where to add the argument but the robot never worked after I added it, so I assume I added it in the wrong spot.

forgetwhatuwant, 2025-10-29:
after the Lerobot released its new version with the pr #1558 been merged, i think u can directly try command like 

```
python -m lerobot.record \
--robot.type=so100_follower \
--robot.port=/dev/tty.XXXX \
--robot.id=my_awesome_follower_arm \
--robot.cameras="{ wrist.right: {type: opencv, index_or_path: 0, width: 1920, height: 1080, fps: 30, fourcc: 'MJPG'}, top: {type: opencv, index_or_path: 1, width: 1920, height: 1080, fps: 30, fourcc: 'MJPG'}}" \
--teleop.type=so100_leader \
--teleop.port=/dev/tty.XXXX \
--teleop.id=my_awesome_leader_arm \
--display_data=true \
--dataset.repo_id=${HF_USER}/record-test \
--dataset.num_episodes=5 \
--dataset.single_task="Stack the yellow birdie"
```