huggingface/lerobot #1738: Raspberry Pi 4 Installation

ccha425, 2025-08-15:
I'm trying to setup lerobot arms on a pi 4 but I kept getting "Illegal instruction" error whenever trying to run any Lerobot-related commands
Followed the official Lerobot installation instructions, which worked on another PC, but can't seem to get hardware control for teleoperating or calibration working. no error code or error log, just a single line saying "Illegal instruction"

lerobot-setup-motors \
    --robot.type=so101_follower \
    --robot.port=/dev/ttyACM0

lerobot-calibrate \
    --robot.type=so101_follower \
    --robot.port=/dev/ttyACM0
    --robot.id=follower_1

Tried to go through the Python script line by line, and it seems to throw the error when importing the lerobot package:
from lerobot.robots import( # noqa:F401
    Robot,
    RobotConfig,
    hope_jr,
    koch_follower,
    lekiwi,
    make_robot_from_config,
    so100_follower,
    so101_follower,
)

Hardware:
Raspberry Pi 4B from OKDO

OS:
64-bit Raspberry OS, Ubuntu 22.04, 20.04, 18.04

Conda install
Tried both 
wget "https://github.com/conda-forge/miniforge/releases/latest/download/Miniforge3-$(uname)-$(uname -m).sh"
and the aarch64
https://www.anaconda.com/docs/getting-started/miniconda/install
wget https://repo.anaconda.com/miniconda/Miniconda3-latest-Linux-aarch64.sh

Installation and activation all worked
Then installed from huggingface lerobot source
pip install -e .
pip install -e ".[feetech]"

Also tried clean install using pypi
pip install lerobot
pip install 'lerobot[feetech]'

pham-tuan-binh, 2025-08-15:
I have the same problem on Pi Zero 2W. I tested every dependency that lerobot pulled in and the culprit seems to be "datasets". Have no fix for this atm though/

LixingZHANG96, 2025-08-16:
Hi there, I am having exactly the same issue here. Any solution found so far?

LixingZHANG96, 2025-08-19:
@pham-tuan-binh Confirmed solution also from my side. You are a hero!

jamminc, 2025-10-04:
@pham-tuan-binh Thank you, solution worked for me too!!

jadechoghari, 2025-10-17:
Closing this issue as it has been resolved.
Thank you for using LeRobot!
Feel free to reopen it if you encounter any related issues.