huggingface/lerobot #1763: The best camera setup for fine-tuning SmolVLA with so101

yuanlin2004, 2025-08-19:
I am fine-tuning SmolVLA for the pick-and-place task for my so101. What is the best camera setup? And what is the order to specify them for record.py?

I find the following pieces of information, which do not give a coherent answer. 

1. The SmolVLA [paper](https://arxiv.org/pdf/2506.01844) says 'prioritizing top, wrist, and side perspectives–and renamed them as OBS_IMAGE_1, OBS_IMAGE_2, and OBS_IMAGE_3'. 

2. This dataset from Huggingface for so101 [lerobot/svla\_so101\_pickplace · Datasets at Hugging Face](https://huggingface.co/datasets/lerobot/svla_so101_pickplace) uses 'up' and 'side'.

3. This dataset from Huggingface for so100 [lerobot/svla\_so101\_pickplace · Datasets at Hugging Face](https://huggingface.co/datasets/lerobot/svla_so101_pickplace) uses 'top' and 'wrist'.
 
4. The SmoVLA code ([prepare_images] (https://github.com/huggingface/lerobot/blob/55198de096f46a8e0447a8795129dd9ee84c088c/src/lerobot/policies/smolvla/modeling_smolvla.py#L479) and [embed_prefix] (https://github.com/huggingface/lerobot/blob/55198de096f46a8e0447a8795129dd9ee84c088c/src/lerobot/policies/smolvla/modeling_smolvla.py#L692) ) simply concatenates the image tokens from all the cameras (in the order specified in the command line argument of record.py) together with language tokens and state tokens as the input to SmolVLM. It does not matter how many cameras are used, as long as there is at least one. The names of the cameras do not matter either. There seems no clear way to tell from the embedding input which token is for which image or text. 
 
Is keeping the semantics of the embedding input in the fine-tuning data consistent with that in the pre-training data important for fine-tuning?

In my experiment, my fine-tuning [dataset](https://huggingface.co/spaces/lerobot/visualize_dataset?path=%2FTangyKiwi%2Frecord-test%2Fepisode_0) used 'top', 'front' and 'wrist' cameras and in that order. I found I got better result during inference when I keep the same order (i.e. `--robot.cameras="{ top: {..},  front: {..},  wrist: {..}}`, and worse result if different order (e.g. `--robot.cameras="{ top: {..},  wrist: {..},  front: {..}}`). But I do not know my camera order is consistent with the pre-training data. 

Could someone familiar with SmoVLA and code give an authoritative answer?

yuanlin2004, 2025-12-10:
I used top/front/wrist. The performance was not good.

mcres, 2025-12-19:
> The robot arm always got stuck at a fixed position, is this the expected behavior?

@MyLovelyAxe I also reproduced the same behavior you described, using the model without fine-tuning. Something similar was also reproduced by @alexcbb [here](https://huggingface.co/blog/Beegbrain/daily-robotics-june-2#a-zero-shot-tests-early-friction).

I think this comment will help: https://github.com/huggingface/lerobot/issues/2221#issuecomment-3413221473

MyLovelyAxe, 2026-01-01:
> > The robot arm always got stuck at a fixed position, is this the expected behavior?
> 
> [@MyLovelyAxe](https://github.com/MyLovelyAxe) I also reproduced the same behavior you described, using the model without fine-tuning. Something similar was also reproduced by [@alexcbb](https://github.com/alexcbb) [here](https://huggingface.co/blog/Beegbrain/daily-robotics-june-2#a-zero-shot-tests-early-friction).
> 
> I think this comment will help: [#2221 (comment)](https://github.com/huggingface/lerobot/issues/2221#issuecomment-3413221473)

I think I found the reason of the issue that the robot arm always got stuck at the same position and kept wobbling. If SmolVLA base model is loaded from pre-trained model by:

```
from lerobot.policies.smolvla.modeling_smolvla import SmolVLAPolicy
model = SmolVLAPolicy.from_pretrained("lerobot/smolvla_base")
```

then according to the source code, the observation input is not normalized and action output is not denormalized. This is caused by a mis-match of label names in processor `safetensors` file and dataset structure. Check source code in [lerobot/src/lerobot/processor/normalize_processor.py](https://github.com/huggingface/lerobot/blob/9701b9c2735a989a6f07b5b3128db73408245c72/src/lerobot/processor/normalize_processor.py#L306):

```
        if norm_mode == NormalizationMode.IDENTITY or key not in self._tensor_stats:
            return tensor
```

The variable `key` here is string `"action"` defined in this file, which is also the dataset feature defined in `config.json`. But the `self._tensor_stats` is loaded from these `safetensors` files (since the policy was loaded by functioin `from_pretrained()`), check your downloaded pretrained model here:

```
~/.cache/huggingface/hub/models--lerobot--smolvla_base/snapshots/4d2f2b37fa245361ef1efe6d91ce96b8bd4af511/policy_postprocessor_step_0_unnormalizer_processor.safetensors
~/.cache/huggingface/hub/models--lerobot--smolvla_base/snapshots/4d2f2b37fa245361ef1efe6d91ce96b8bd4af511/policy_preprocessor_step_5_normalizer_processor.safetensors
```

These 2 `safetensors` are the ones got loaded into the `self._tensor_stats` above, if you load them with:

```
from safetensors.torch import load_file
from pathlib import Path
folder = Path("/home/hardli/.cache/huggingface/hub/models--lerobot--smolvla_base/snapshots/4d2f2b37fa245361ef1efe6d91ce96b8bd4af511")
postprocessor_safetensor = "policy_postprocessor_step_0_unnormalizer_processor.safetensors"
preprocessor_safetensor = "policy_preprocessor_step_5_normalizer_processor.safetensors"
post_processor = load_file(folder / postprocessor_safetensor)
pre_processor = load_file(folder / preprocessor_safetensor)
```

You will find the key names don't have a `action`:

```
{'so100-blue.buffer.action.mean': tensor([   1.4212,  125.7205,  125.4193,   63.7832, -106.5401,    3.3485]),
 'so100-blue.buffer.action.std': tensor([14.0820, 39.4314, 18.2381, 23.4898, 38.1112,  5.0483]),
 'so100-red.buffer.action.mean': tensor([   2.4387,  124.9379,  123.5578,   66.9926, -103.8768,    2.8912]),
 'so100-red.buffer.action.std': tensor([14.2992, 42.7722, 18.8763, 21.6519, 33.5341,  4.4780]),
 'so100.buffer.action.mean': tensor([  1.5959, 119.9442, 109.7698,  56.7055, -27.4227,  12.0028]),
 'so100.buffer.action.std': tensor([26.3918, 52.4114, 49.8538, 36.9983, 59.3600, 19.0400])}
```

and according to this if-condition mentioend above:

```
        if norm_mode == NormalizationMode.IDENTITY or key not in self._tensor_stats:
            return tensor
```

the `key not in self._tensor_stats` is always `True`, and the input `tensor` is just returned without any change. This means input is directly sent into policy without normalization, and output was sent out from policy without denormalization. Since the poliy itself was trained by normalized data, if without normalization, any input is not the same distribution that the policy expects. This causes the behavior that robot arm always wobbles at the same position, because the input and output are not correctly dealt with. However, as you see, this mismatch comes from:

- pre-trained safetensors for processors
- dataset feature (i.e. key name `"action"`) that SmolVLA was trained on

but since that if-condition just lets the input tensor returned without raising any error, even if it doesn't find correct key name in `self._tensor_stats`, it is hard to find whether the input and out are really normalized.

You can fix this temporarily by changing the key names in safetensors or dataset structure to let it match, afterwards, the robot should not get stuck anymore, even if zero-shot performance is not ideal, it looks more reasonable. But probably this still needs an official fix from Lerobot repo.