huggingface/lerobot #1774: Finetune smolvla with vision encoder

THU-yancow, 2025-08-22:
### System Info

```Shell
- `lerobot` version: 0.1.0
- Platform: Linux-6.8.0-65-generic-x86_64-with-glibc2.35
- Python version: 3.10.18
- Huggingface_hub version: 0.33.4
- Dataset version: 3.6.0
- Numpy version: 2.2.6
- PyTorch version (GPU?): 2.7.1+cu126 (True)
- Cuda version: 12060
- Using GPU in script?: <fill in>
```

### Information

- [ ] One of the scripts in the examples/ folder of LeRobot
- [x] My own task or dataset (give details below)

### Reproduction

nothing

### Expected behavior

I found that when attempting to fine-tune the model to grasp objects of different colors but identical shapes, it consistently grasped the wrong object. I found that the output feature differences from the VLM for the same image, such as “grasp the green duck into the box” versus “grasp the yellow duck into the box,” were nearly zero. Is it possible that the VLM has weak color differentiation capabilities? Can the official support fine-tuning the visual encoder together?

boysack, 2026-02-13:
In the original paper, under Experimental Setup->Real-world tasks section ([SmolVLA](https://arxiv.org/pdf/2506.01844#subsection.4.1)), one of the experiments used for evaluation (the sorting one) is checking exactly for color differentiation capabilities. Unfortunately, the only result provided is an aggregation of all the experiments.

To unlock visual encoder fine-tuning, you could follow the official guide [Bring Your Own Policies](https://huggingface.co/docs/lerobot/bring_your_own_policies) in order to create a new policy from the original SmolVLA.
The simplest way to achieve this is to create a new policy by inheriting the original configuration, modeling and processor SmolVLA classes.

If you want to fine-tune just the vision_model, create a new set_requires_grad in the modeling file and use it in the initialization:
```
class SmolVLAUnfrozenPolicy(SmolVLAPolicy):
  config_class = SmolVLAUnfrozenConfig
  name = "smolvla_unfrozen"

  def __init__(self, config, **kwargs):
    super().__init__(config, **kwargs)
    self.set_requires_grad()

  def set_requires_grad(self):
    if self.config.fine_tune_vision_encoder:
      for params in self.model.vlm_with_expert.get_vlm_model().vision_model.parameters():
        params.requires_grad = True
...
```

Moreover, it could be useful to set the learning rate of the vision_model to something smaller than the rest of the network. To achieve this:
```
@PreTrainedConfig.register_subclass("smolvla_unfrozen")
@dataclass
class SmoLVLAUnfrozenConfig(SmoLVLAConfig) :
  fine_tune_vision_encoder: bool = True
  vision_lr_multiplier: float = 0.1
...
```
```
class SmolVLAUnfrozenPolicy(SmolVLAPolicy):
...
  def get_optim_params(self) -> dict:
    if not self.config.fine_tune_vision_encoder:
      return super().get_optim_params()
    vision_params = []
    params = []    
    for name, param in self.named_parameters():
      if not param.requires_grad:
        continue
      if "vision_model" in name:
        vision_params.append(param)
      else:
        params.append(param)
    return [
        {"params": vision_params, "lr": self.config.optimizer_lr * self.config.vision_lr_multiplier},
        {"params": params, "lr": self.config.optimizer_lr},
    ]
```
Since it's not easily possible to override the original lerobot policies without changing the source code, my suggestion is to download pretrained weights locally (e.g. `lerobot/smolvla_base`) and then editing accordingly to the needs the `config.json` (use [override_policy.py](https://github.com/user-attachments/files/25288983/override_policy.py) to do this).

Consider that, while it's worth to try, fine-tuning in this way could be insufficient: working with the vision model weights could lead the network to catastrophically forget the general knowledge learned during pretraining, even using a lower learning rate for that.
For this reason, PEFT methods, such as LoRa or QLoRa, are worth to try, since they are proven to mitigate forgetting, while achieving comparable performance with full fine-tuning.

michaellinhq, 2026-02-13:
Hi, I reviewed 'Finetune smolvla with vision encoder' and can start immediately. I can deliver a first usable version quickly and iterate based on your feedback. If this scope still matches your priority, I can proceed right away.