huggingface/lerobot #1819: 关于Hirserl的复现：

bxsytt, 2025-08-30:
真诚发问，我用so101主臂控制so100从臂：
1.首先运行find_joint_limits.py生成两个json文件，并且将终端输出的min和max写入到end_effector_bounds，但是感觉min和max的差距很小，请问是我校准操作有问题吗？
2.校准过程有前辈告诉我一下吗，我运行python -m lerobot.scripts.rl.gym_manipulator --config_path src/lerobot/configs/env_config_so100.json
，so100会急速运动到一个位置，感觉应该让机械臂运动到目标抓取位置才对吧？

  "type": "gym_manipulator",
  "robot": {
    "type": "so100_follower_end_effector",
    "port": "/dev/ttyACM0",
    "id": "black_so100_follower",
    "use_degrees": false,
    "disable_torque_on_disconnect": true,
    "cameras": {
      "main": {
        "type": "opencv",
        "index_or_path": "/dev/video4",
        "width": 640,
        "height": 480,
        "fps": 30
      },
      "secondary": {
        "type": "opencv",
        "index_or_path": "/dev/video2",
        "width": 640,
        "height": 480,
        "fps": 30
      }
    },
    "max_gripper_pos": 50,
    "end_effector_step_sizes": {
      "x": 0.02,
      "y": 0.02,
      "z": 0.02
    }
  },
  "teleop": {
    "type": "so101_leader",
    "port": "/dev/ttyACM1",
    "id": "blue_so101_leader",
    "use_degrees": false,
    "end_effector_bounds": {
      "min": [0.071, 0.0013, 0.4202],
      "max": [0.0721, 0.0026, 0.4205]
    }
  },
  "wrapper": {
    "control_mode": "leader",
    "use_gripper": true,
    "display_cameras": false,
    "control_time_s": 20.0,
    "reset_time_s": 5.0,
    "gripper_quantization_threshold": 0.8,
    "gripper_penalty": 0.0,
    "gripper_penalty_in_reward": false,
    "crop_params_dict": null
  },
  "mode": "record",
  "repo_id": "btt_herl/pick_and_lift_green_cube",
  "dataset_root": "/home/xbot/btt2204/lerobot_smolvla/datasets/hilserl_pick_cube",
  "task": "pick_and_lift_green_cube",
  "num_episodes": 15,
  "episode": 0,
  "push_to_hub": true,
  "fps": 10,
  "number_of_steps_after_success": 0
}

Hukongtao, 2025-09-01:
大佬直接买的机器人嘛？

bxsytt, 2025-09-03:
> 大佬直接买的机器人嘛？

对的，目前在复现，遇到很多问题