huggingface/lerobot #1820: Suggest to install conda via miniforge instead of miniconda?

traversaro, 2025-08-31:
Currently, in `lerobot` instructions, the users are suggested to install conda via the miniconda installer. However, I am afraid that this may problematic for two main reasons:

* **Legal challenges in using `defaults` conda channels**: The `miniconda` installer install itself and by default create environments using the `defaults` conda channel, and according to the Anaconda Inc. their term of service is not free to use for all users (see https://www.anaconda.com/blog/is-conda-free and https://prefix.dev/blog/towards_a_vendor_lock_in_free_conda_experience). Indeed, several university and companies (see https://www.theregister.com/2024/08/08/anaconda_puts_the_squeeze_on/ and https://thehftguy.com/2025/04/07/anaconda-inc-has-entered-litigation-against-non-paying-user-of-anaconda-alibaba-intel-dell-airbus/) had legal problems as they used the `defaults` conda channel without being aware of the ToS. A simple alternative to avoid any kind of legal problem is to only use the `conda-forge` community channel, and so use an installer that only install packages from the `conda-forge` channel (including conda itself) like [`miniforge`](https://github.com/conda-forge/miniforge)
* **Technical challenges in mixing `defaults` and `conda-forge` conda channels**: even if lerobot suggest to install conda via miniconda (and hence creating environments with packages from the `defaults` channels), then the instructions suggest to install packages from the `conda-forge` channel, see for example https://github.com/huggingface/lerobot/blame/882c80d446a63a44868c67ae535467af32ce0e80/docs/source/installation.mdx#L20 . As of 2025, **mixing `conda-forge` and `default` packages is explicitly not supported**, see conda-forge docs in https://conda-forge.org/docs/user/transitioning_from_defaults/ , so mixing `conda-forge` and `defaults` can lead to crash in the future (if this never happened, I guess is just due to lucky combination of packages used that did not result in any kind of ABI incompatibility, but this may change in the future).

For these two reasons, I think it could make sense to just suggest to use `miniforge` instead of `miniconda` in lerobot documentation.

traversaro, 2025-08-31:
To be a bit more concrete, the suggested modification to fix this issue are proposed in https://github.com/huggingface/lerobot/pull/1821 .

imstevenpmwork, 2025-10-17:
Thanks for the detailed explanation. I will take a look at this after the next release, thanks !

traversaro, 2025-10-17:
Thanks @imstevenpmwork !

> * As of 2025, **mixing `conda-forge` and `default` packages is explicitly not supported**, see conda-forge docs in https://conda-forge.org/docs/user/transitioning_from_defaults/ , so mixing [...]

To add more concrete details to this, there have been at least 3 crashes on the torchcodec side due to mixing `defaults` and `conda-forge`:
* https://github.com/meta-pytorch/torchcodec/issues/518, fix https://github.com/meta-pytorch/torchcodec/pull/658
* https://github.com/meta-pytorch/torchcodec/issues/974, fix https://github.com/meta-pytorch/torchcodec/pull/975
* https://github.com/meta-pytorch/torchcodec/issues/951, fix https://github.com/meta-pytorch/torchcodec/pull/973

imstevenpmwork, 2025-10-17:
Concerning, I agree this needs to be solved. Thanks for the heads up !