huggingface/lerobot #1834: Questions about asynchronous inference

godnpeter, 2025-09-01:
Hi, I'm currently trying to evaluate a fine-tuned  SmolVLA checkpoint for a SO-100 robot arm. 
I've tried evaluating my policy both with `lerobot.record` and `asynchronous inference` and what I'm experiencing is that although `lerobot.record` works fully ondevice (MAC with M3 chip), it works way better than using `asynchronous inference`, which is currently using a A100 as its inference GPU.

The Avg FPS I currently observe for the asynchronous version seems to be way lower than I first anticipated. I'm curious if there are others who also experiencing the same phenomenon (where on-device actually works better than asynchronous inference).

```
INFO 2025-09-01 14:56:14 y_server.py:223 Action chunk #3384 generated | Total time: 320.03ms
INFO 2025-09-01 14:56:15 y_server.py:175 Received observation #3385 | Avg FPS: 0.43 | Target: 30.00 | One-way latency: 94.69ms
INFO 2025-09-01 14:56:15 y_server.py:205 Running inference for observation #3385 (must_go: False)
INFO 2025-09-01 14:56:15 y_server.py:355 Observation 3385 |Inference time: 307.54ms
INFO 2025-09-01 14:56:15 y_server.py:223 Action chunk #3385 generated | Total time: 311.20ms
INFO 2025-09-01 14:56:16 ort/utils.py:74 <Logger policy_server (NOTSET)> Starting receiver
INFO 2025-09-01 14:56:17 y_server.py:175 Received observation #3409 | Avg FPS: 0.43 | Target: 30.00 | One-way latency: 88.89ms
INFO 2025-09-01 14:56:17 y_server.py:205 Running inference for observation #3409 (must_go: False)
INFO 2025-09-01 14:56:17 ort/utils.py:74 <Logger policy_server (NOTSET)> Starting receiver
INFO 2025-09-01 14:56:17 y_server.py:355 Observation 3409 |Inference time: 312.51ms
INFO 2025-09-01 14:56:17 y_server.py:223 Action chunk #3409 generated | Total time: 316.29ms
INFO 2025-09-01 14:56:19 y_server.py:175 Received observation #3410 | Avg FPS: 0.43 | Target: 30.00 | One-way latency: 91.23ms
```

helper2424, 2025-09-01:
Looks interesting, could you turn on debug level for logging - so we can see more details?

Also, there is another PR with small fixed for transport - https://github.com/huggingface/lerobot/pull/1486/files. U can try it. I haven't finished it because stuck on providing RTC, but hope will back to it soon.

jadechoghari, 2025-09-01:
cc @fracapuano for async stuff

wjpoom, 2025-11-17:
Hi, we also meet the same issue, when use async inference, the fps is very slow. I checked the log and find de-serialization is the  most time-consuming process, which takes about 6 seconds per observation. Is there any advice? thanks