huggingface/lerobot #1875: ValueError: One or several timestamps unexpectedly violate the tolerance inside episode range. This might be due to synchronization issues during data collection.

cuijianaaa, 2025-09-05:
### System Info

```Shell
- `lerobot` version: 0.3.4
- Platform: Linux-6.8.0-79-generic-x86_64-with-glibc2.35
- Python version: 3.10.18
- Huggingface_hub version: 0.34.4
- Dataset version: 3.6.0
- Numpy version: 2.2.6
- PyTorch version (GPU?): 2.7.1+cu126 (True)
- Cuda version: 12060
- Using GPU in script?: <fill in>
```

### Information

- [x] One of the scripts in the examples/ folder of LeRobot
- [ ] My own task or dataset (give details below)

### Reproduction

I run the SmolVLA example command in https://huggingface.co/blog/smolvla，This instruction command about SmolVLA seems to be outdated and cannot be run according to the given command
```
lerobot-train \
  --policy.push_to_hub=false \
  --policy.path=lerobot/smolvla_base \
  --dataset.repo_id=lerobot/svla_so100_stacking \
  --batch_size=64 \
  --steps=20000  # 10% of training budget

```

```
file-000.mp4: 100%|█████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████| 186M/186M [02:43<00:00, 1.14MB/s]
file-000.mp4: 100%|█████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████| 404M/404M [04:09<00:00, 1.62MB/s]
Fetching 180 files: 100%|███████████████████████████████████████████████████████████████████████████████████████| 180/180 [09:06<00:00,  3.04s/it]                                  | 0.00/186M [00:00<?, ?B/s]
Resolving data files: 100%|████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████| 57/57 [00:00<00:00, 581691.80it/s]
Downloading data: 100%|█████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████| 57/57 [00:00<00:00, 195482.69files/s]
Generating train split: 45912 examples [00:00, 528211.16 examples/s]███████████████████████████████████████████████████████████████████████████████████████████████████████| 3.17M/3.17M [00:08<00:00, 358kB/s]
Traceback (most recent call last):████████████████████████████████████████████████████████▍                                                                                 | 189M/404M [02:33<02:35, 1.38MB/s]
  File "/home/cj/miniconda3/envs/lerobot/bin/lerobot-train", line 7, in <module>████████████████████████████████████████████████████████████████████████████████████████████| 404M/404M [04:09<00:00, 2.51MB/s]
    sys.exit(main())████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████| 186M/186M [02:43<00:00, 1.40MB/s]
  File "/home/cj/repos/lerobot/src/lerobot/scripts/train.py", line 291, in main
    train()
  File "/home/cj/repos/lerobot/src/lerobot/configs/parser.py", line 225, in wrapper_inner
    response = fn(cfg, *args, **kwargs)
  File "/home/cj/repos/lerobot/src/lerobot/scripts/train.py", line 128, in train
    dataset = make_dataset(cfg)
  File "/home/cj/repos/lerobot/src/lerobot/datasets/factory.py", line 90, in make_dataset
    dataset = LeRobotDataset(
  File "/home/cj/repos/lerobot/src/lerobot/datasets/lerobot_dataset.py", line 492, in __init__
    check_timestamps_sync(timestamps, episode_indices, ep_data_index_np, self.fps, self.tolerance_s)
  File "/home/cj/repos/lerobot/src/lerobot/datasets/utils.py", line 574, in check_timestamps_sync
    raise ValueError(
ValueError: One or several timestamps unexpectedly violate the tolerance inside episode range.
                This might be due to synchronization issues during data collection.
                
[{'diff': np.float32(-4.6666665),
  'episode_index': 55,
  'timestamps': [np.float32(4.6666665), np.float32(0.0)]},
 {'diff': np.float32(-14.866667),
  'episode_index': 0,
  'timestamps': [np.float32(14.866667), np.float32(0.0)]},
 {'diff': np.float32(-13.833333),
  'episode_index': 1,
  'timestamps': [np.float32(13.833333), np.float32(0.0)]},
 {'diff': np.float32(-14.3),
  'episode_index': 2,
  'timestamps': [np.float32(14.3), np.float32(0.0)]},
 {'diff': np.float32(-12.233334),
[... 136 lines cut ...]
  'timestamps': [np.float32(14.966666), np.float32(0.0)]},
 {'diff': np.float32(-17.266666),
  'episode_index': 49,
  'timestamps': [np.float32(17.266666), np.float32(0.0)]},
 {'diff': np.float32(-14.366667),
  'episode_index': 50,
  'timestamps': [np.float32(14.366667), np.float32(0.0)]},
 {'diff': np.float32(-11.433333),
  'episode_index': 51,
  'timestamps': [np.float32(11.433333), np.float32(0.0)]},
 {'diff': np.float32(-12.533334),
  'episode_index': 52,
  'timestamps': [np.float32(12.533334), np.float32(0.0)]},
 {'diff': np.float32(-13.766666),
  'episode_index': 53,
  'timestamps': [np.float32(13.766666), np.float32(0.0)]},
 {'diff': np.float32(-14.666667),
  'episode_index': 54,
  'timestamps': [np.float32(14.666667), np.float32(0.0)]}]

```

### Expected behavior

Normal training without errors

johnnycrab, 2025-09-13:
@brickerino As far as I understand it, it doesn't as long as all episodes are there. In `lerobot_dataset.py` it makes this check within a "try-except download"-block in line 480: `assert all((self.root / fpath).is_file() for fpath in self.get_episodes_file_paths())`. 
I just tried it: when you only delete `file-000.parquet`, all is fine. If you delete an episode file, it will download not only the missing episode, but also `file-000.parquet` again. 🤷‍♀️

rns0869, 2025-09-20:
I am having the same problem, when I try to train a Model. 

Can someone please suggest a quick fix?