huggingface/lerobot #1899: Has anyone tried to export the smolvla as onnx model for deployment?

TankerLee, 2025-09-09:
I have tried to test the trained smolvla model on my PC, it works. I want now to deploy the smolvla on our target board. 

I looked into the model structure of smolvla, for the vision-encoder and language embedding parts I can refer to the smolvlm and export them as tow onnx models. I think the robot state embedding also needs to be considered to export as a new onnx model.

The most important part of smolvla inference, i met several issues and have no good idea how to export it as a onnx model.l.

Has anyone tried and successfully exported the smolvla as onnx models for deployment? Thanks!

1104992692, 2025-10-04:
Is Your board a  fpga ?

rylinjames, 2026-05-06:
solved this -- it's in reflex-vla (github.com/FastCrest/reflex-vla).

the hard part you identified (the flow diffusion/action expert) took a lot of work to get right. the main issues were: broken kv cache wiring in the VLM prefix (past_key_values not being passed correctly), wrong sinusoidal timestep embedding (missing 2*pi factor), image dims (AutoProcessor returns 5D [B,N,3,H,W] which broke the ONNX trace), and RoPE theta wrong (10000 vs 100000 for SmolLM2). we export the full denoising loop unrolled as a monolithic ONNX so the TRT engine is a single graph.

`reflex export --model smolvla` should get you there. cos=+1.0 parity against pytorch verified.