huggingface/lerobot #1959: write_image() is slow due to default compress_level=6

playertr, 2025-09-16:
### System Info

```Shell
- `lerobot` version: 0.3.4
- Platform: Linux-6.10.14-linuxkit-aarch64-with-glibc2.39
- Python version: 3.12.3
- Huggingface_hub version: 0.34.5
- Dataset version: 3.6.0
- Numpy version: 2.2.6
- PyTorch version (GPU?): 2.7.1+cpu (False)
- Cuda version: N/A
- Using GPU in script?: <fill in>
```

### Information

- [ ] One of the scripts in the examples/ folder of LeRobot
- [x] My own task or dataset (give details below)

### Reproduction

To reproduce this bug, we 
- load real robot video from an existing dataset
- write ten frames of video into a dataset and see how long it takes
- monkeypatch in a local change adding the `compress_level` kwarg to `PIL.save`
- observe that writing video into the dataset becomes much faster

You can reproduce this with the Docker setup below. To do this, copy the Dockerfile and `repro.py` from the expandible sections below into separate files in a local directory. Then, run:
```
docker build . -t lerobot_repro
docker run --rm -it \
  -v "$PWD"/lerobot:/src/lerobot \
  -v "$PWD"/repro.py:/src/repro.py \
  -v "$PWD"/.cache/huggingface:/root/.cache/huggingface \
  lerobot_repro 
  bash -c "cd /src/ && source venv/bin/activate && python repro.py"
```

<details>
  <summary>Click to expand `Dockerfile`.</summary>
    ```

    FROM ubuntu
    
    RUN apt-get update && apt-get install -y \
        python3 \
        python3-pip \
        python3-venv \
        && rm -rf /var/lib/apt/lists/*
    
    RUN mkdir /src
    
    COPY lerobot /src/lerobot
    
    RUN cd /src \
        && python3 -m venv venv \
        && . venv/bin/activate \
        && python3 -m pip install -e /src/lerobot
    ```
</details>

<details>
  <summary>Click to expand `repro.py`. </summary>

    ```

    import lerobot.datasets.lerobot_dataset
    from lerobot.datasets.lerobot_dataset import LeRobotDataset
    import time
    from pathlib import Path
[... 114 lines cut ...]
Svt[info]: -------------------------------------------
Svt[info]: SVT [config]: main profile   tier (auto)     level (auto)
Svt[info]: SVT [config]: width / height / fps numerator / fps denominator               : 1280 / 720 / 10 / 1
Svt[info]: SVT [config]: bit-depth / color format                                       : 8 / YUV420
Svt[info]: SVT [config]: preset / tune / pred struct                                    : 8 / PSNR / random access
Svt[info]: SVT [config]: gop size / mini-gop size / key-frame type                      : 2 / 32 / key frame
Svt[info]: SVT [config]: BRC mode / rate factor                                         : CRF / 30 
Svt[info]: SVT [config]: AQ mode / variance boost                                       : 2 / 0
Svt[info]: SVT [config]: sharpness / luminance-based QP bias                    : 0 / 0
Svt[info]: Svt[info]: -------------------------------------------
Generating train split: 1 examples [00:00, 163.82 examples/s]
Wrote a 10-frame dataset in 1.45 seconds.)
Dataset size on disk: 0.28 MB
-----------------------------------------------------------------------------------------------------=
```

There was a significant (50%) speedup for the entire dataset writing from disabling PNG compression, from 2.45 to 1.45 seconds. This corresponds to a huge compute time savings when writing long, high-resolution video episodes.

### Expected behavior

Ideally, we would expose the `compress_level` argument so that the user can specify their desired compression level without requiring a monkeypatch. This may involve adding a keyword argument to write_image and exposing it through the AsyncImageWriter class and the LeRobotDataset class

For use cases like in `repro.py` where we are eventually saving an MP4 video anyway, there is very little incentive to compress the PNG images when they are temporarily stored. The compression time in `write_image` has a dramatic effect: I've observed close to 10X decrease in runtime by using `compress_level=0`. I suspect that reducing the write time by exposing the `compress_level` argument would be a better solution for most people than increasing write parallelism through `image_writer_processes` and `image_writer_threads`.

I didn't submit these changes in a PR because the maintainers might have feedback on the best way to expose this parameter.

CarolinePascal, 2025-09-29:
Hi @playertr,

Very nice catch ! I took the liberty to slightly modify your `repro.py` script to include the three errors we used in our encoding-decoding benchmark (mean_squared_error, peak_signal_noise_ratio and structural_similarity) to assess if the quality of the images was somewhat impacted by the compression level.

Interestingly, the errors stay the same regardless of the compression level ! I'm going to reproduce the experiment with another dataset tomorrow and post the results here. 

If this observation extends to other datasets, I think we can safely move on by setting `compress_level = 0` by default in the codebase. We will work on a PR to make this parameter as well as the the encoding/decoding parameters in the near future ;)

Best,

Caroline.

CarolinePascal, 2025-09-30:
Hi @playertr,

As promised here are the results from my small investigation. 

<details>
<summary>Click to expand the modified `repro.py`</summary>
```
import lerobot.datasets.lerobot_dataset
from lerobot.datasets.lerobot_dataset import LeRobotDataset
import time
from pathlib import Path
import numpy as np
import shutil
from importlib import reload
from skimage.metrics import mean_squared_error, peak_signal_noise_ratio, structural_similarity

DS_ROOT = Path("/tmp/test")

def size_of_dir(dir: Path) -> int:
    return sum(f.stat().st_size for f in dir.glob('**/*') if f.is_file())

def benchmark_write_time_and_dataset_size(images: list[np.ndarray]) -> None: 
    images_shape = images[0].shape
    ds = LeRobotDataset.create(
        repo_id = "test",
        root= DS_ROOT,
        fps = 30,
        features={
            "image": {"dtype": "video", "shape": images_shape},
        },
        use_videos=True
    )

    tic = time.perf_counter()
    for i in range(100):
        ds.add_frame({"image": images[i], "task": "test_task"})
    ds.save_episode()
    toc = time.perf_counter()

    duration = toc - tic
    size = size_of_dir(DS_ROOT)

    mse_errors = []
    psnr_errors = []
    ssim_errors = []

    for i, original_frame in enumerate(images):
        dataset_frame_np = ds[i]['image'].numpy()
        original_frame_np = original_frame.numpy()
        mse_errors.append(mean_squared_error(dataset_frame_np, original_frame_np))
        psnr_errors.append(peak_signal_noise_ratio(dataset_frame_np, original_frame_np, data_range=1.0))
        ssim_errors.append(structural_similarity(dataset_frame_np, original_frame_np, data_range=1.0, channel_axis=0))

    avg_mse_errors = np.array(mse_errors).mean()
    avg_psnr_errors = np.array(psnr_errors).mean()
    avg_ssim_errors = np.array(ssim_errors).mean()

    return duration, size, avg_mse_errors, avg_psnr_errors, avg_ssim_errors

## Load high-res images from an existing dataset
repo_id = 'CarolinePascal/kitchen'
input_ds = LeRobotDataset(repo_id, episodes=[0])
camera_keys = input_ds.meta.camera_keys
if len(camera_keys) == 0:
    raise ValueError("No camera keys found in the dataset")
camera_key = camera_keys[0]

images = []
for i in range(100): 
    frame = input_ds[i]
    images.append(frame[camera_key])

durations = []
sizes = []

mse_errors = []
psnr_errors = []
ssim_errors = []

import tqdm

for compress_level in tqdm.tqdm(range(10), desc="Compress levels"):

    ## Monkeypatch in a compress_level argument
    from lerobot.datasets import image_writer 
    from lerobot.datasets.image_writer import image_array_to_pil_image
    import PIL

    def write_image_fast(image: np.ndarray | PIL.Image.Image, fpath: Path):
        """Identical to upstream except for compress_level arg."""
        try:
            if isinstance(image, np.ndarray):
                img = image_array_to_pil_image(image)
            elif isinstance(image, PIL.Image.Image):
                img = image
            else:
                raise TypeError(f"Unsupported image type: {type(image)}")
            img.save(fpath, compress_level=compress_level)
        except Exception as e:
            print(f"Error writing image {fpath}: {e}")

    image_writer.write_image = write_image_fast
    reload(lerobot.datasets.lerobot_dataset)

    ## With these changes, benchmark the time and final size.
    ## Print how long it takes to write a dataset with 100 images
    duration, size, mse_error, psnr_error, ssim_error = benchmark_write_time_and_dataset_size(images)
    durations.append(duration)
    sizes.append(size/(1024**2))
    mse_errors.append(mse_error)
    psnr_errors.append(psnr_error/10)
    ssim_errors.append(ssim_error)

    shutil.rmtree(DS_ROOT)

import plotly.graph_objs as go
from plotly.subplots import make_subplots

# Create subplot structure (3 rows, 1 column)
fig = make_subplots(rows=3, cols=1, subplot_titles=("Duration", "Size", "Errors"))

# First plot: durations
fig.add_trace(go.Scatter(y=durations, mode="lines", name="Duration", legendgroup="1"),
              row=1, col=1)
fig.update_xaxes(title_text="Compress level", row=1, col=1)
fig.update_yaxes(title_text="Duration (s)", row=1, col=1)

# Second plot: sizes
fig.add_trace(go.Scatter(y=sizes, mode="lines", name="Size", legendgroup="2"),
              row=2, col=1)
fig.update_xaxes(title_text="Compress level", row=2, col=1)
fig.update_yaxes(title_text="Size (MB)", row=2, col=1)

# Third plot: errors
fig.add_trace(go.Scatter(y=mse_errors, mode="lines", name="MSE", legendgroup="3"),
              row=3, col=1)
fig.add_trace(go.Scatter(y=psnr_errors, mode="lines", name="PSNR/10", legendgroup="3"),
              row=3, col=1)
fig.add_trace(go.Scatter(y=ssim_errors, mode="lines", name="SSIM", legendgroup="3"),
              row=3, col=1)
fig.update_xaxes(title_text="Compress level", row=3, col=1)
fig.update_yaxes(title_text="Error level", row=3, col=1)

# Layout
fig.update_layout(height=900, width=700, showlegend=True, legend_tracegroupgap=263)

fig.show()
```
</details>

Here are the obtained stats on `lerobot/unitreeh1_fold_clothes`:
[image]
And on `CarolinePascal/kitchen`:
[image]

These curves are so flat it is even worrying somehow...

Anyway, now that we made sure `compress_level = 0` has no (significant) impact on the image quality, you can safely open a PR setting this parameter as default.

Thanks again for opening this issue !

Best,

Caroline.

P.S. I checked the `optimize` parameter as well, and the results are similar : the save time increases, but the size/errors remain the same.