huggingface/lerobot #1966: SO101FollowerEndEffector?

cashlo, 2025-09-17:
I am trying to get inverse kinematics to work on my SO-101, and I found SO100FollowerEndEffector but there is no SO101FollowerEndEffector?

I suspect they are interchangeable, but when I use SO100FollowerEndEffector on my SO-101, it want me to recalibrate it, so I just want to make sure before I break anything.

cashlo, 2025-09-18:
It did work but now it jumps around like an excited ostrich after a few moves, the action values seems fine, so looks like its something with the inverse kinematics
```python
#!/usr/bin/env python

import logging
import os
import time

from lerobot.robots import make_robot_from_config, so101_follower
from lerobot.robots.so100_follower.config_so100_follower import SO100FollowerEndEffectorConfig
from lerobot.robots.so100_follower.so100_follower_end_effector import SO100FollowerEndEffector
from lerobot.robots.so101_follower.config_so101_follower import SO101FollowerConfig
from lerobot.robots.so101_follower.so101_follower import SO101Follower
from lerobot.teleoperators.keyboard.configuration_keyboard import KeyboardEndEffectorTeleopConfig
from lerobot.teleoperators.keyboard.teleop_keyboard import KeyboardEndEffectorTeleop

logging.basicConfig(level=logging.INFO)

def main():
    print("Initializing SO101 robot arm and keyboard controller...")

    robot_cfg = SO100FollowerEndEffectorConfig(port="/dev/ttyACM1", id="so_101_follower_1", urdf_path="urdf")
    robot = SO100FollowerEndEffector(config=robot_cfg)
    print(f"robot: {robot}")

    teleop_cfg = KeyboardEndEffectorTeleopConfig(use_gripper=True)
    teleop = KeyboardEndEffectorTeleop(config=teleop_cfg)

    try:
        robot.connect()
        teleop.connect()
        print("\n✅ Robot and keyboard connected successfully!")
        
        print("\n--- Controls ---")
        print("  - Arrow Keys: Move the arm in the X-Y plane")
        print("  - Left/Right Shift: Move the arm up/down (Z-axis)")
        print("  - Left/Right Ctrl: Close/Open the gripper")
        print("  - ESC: Exit the program")
        print("----------------\n")
        print(robot.get_observation())
        time.sleep(2)
        robot.bus.enable_torque()

        while teleop.is_connected and robot.is_connected:
            action_dict = teleop.get_action()

            print(f"action_dict: {action_dict}")
            robot.send_action(action_dict)
            
            motor_currents = robot.bus.sync_read("Present_Current")
            
            os.system('cls' if os.name == 'nt' else 'clear')
            
            print("--- Live Force Data (Motor Currents) ---")
            for motor, current in motor_currents.items():
                bar = '█' * int(current / 2) 
                print(f"  - {motor:<15}: {current:<5} {bar}")
            print("\n(Press ESC to exit)")

            time.sleep(1 / 3)

    except Exception as e:
        logging.error(f"An error occurred: {e}")
        import traceback
        traceback.print_exc()
    finally:
        print("\nDisconnecting from the robot...")
        if robot.is_connected:
            robot.disconnect()
        print("Cleanup complete. Exiting.")

if __name__ == "__main__":
    main()
```

milong26, 2025-10-30:
@imstevenpmwork Thanks for your work!
Could you tell me is the coordination of end-effector position is base of so100? 
And I'm trying to write a code saving both end-effector-posiion and joint-space(default in Lerobot record) in action and state

pkooij, 2026-10-21:
The current end-effector workflow uses the standard follower with kinematics processors. Please continue the URDF/calibration discussion in #2413. Closing this older robot-type question.

closed 2026-10-21 by pkooij