huggingface/lerobot #1978: Is there a best fit model to each sim env？

shs822, 2025-09-19:
I try to train diffusion，smolvla，even pi0 on the aloha with 200k steps, and found that they all perform much worse (with less than 10% success rate) than act policy, why? Did each env task exist a best-fit policy? or there are problems on my training strategy.

shs822, 2025-09-25:
> According to the result of the experiment i'v done, gr00t policy works best as well as smolvla failed at the last step that grab the stuff up, plus , I've fine-tuned pi-0 on 8-a100, but it works not that good.根据我所做的实验结果，gr00t 策略效果最好，而且 smolvla 在抓取东西的最后一步失败了，另外，我在 8-a100 上微调了 pi-0，但它效果不是那么好。

ok，i will try it ，thank you

jadechoghari, 2025-10-17:
This is more of an open research question, closing for now.
Thanks for sharing your experiments and for using LeRobot! 🚀
Check out https://huggingface.co/spaces/lerobot/robot-learning-tutorial for more!

closed 2025-10-17 by jadechoghari