huggingface/lerobot #1991: Pi0 interperates single task as array of 1 character tasks

Skylerwiernik, 2025-09-20:
### System Info

```Shell
Config file 

robot:
  ...
teleop:
  ...
dataset:
  repo_id: [some dataset]
  num_episodes: 50
  single_task: "Pick up the object from the table and place it in the bin"
  push_to_hub: true
```

### Information

- [ ] One of the scripts in the examples/ folder of LeRobot
- [ ] My own task or dataset (give details below)

### Reproduction

```
python -m lerobot.record --config_path ./config.yaml --policy.path=lerobot/pi0
```
Observe
```
  File ".../lerobot/src/lerobot/policies/pi0/modeling_pi0.py", line 672, in embed_prefix
    embs = torch.cat(embs, dim=1)
           ^^^^^^^^^^^^^^^^^^^^^^
RuntimeError: Sizes of tensors must match except in dimension 1. Expected size 1 but got size 57 for tensor number 1 in the list.
```
(Note, 57 is the number of characters in my prompt)

### Expected behavior

The robot completes the task

Skylerwiernik, 2025-09-20:
Looks like this was fixed with the new pipeline system in #1452

closed 2025-09-20 by Skylerwiernik

AdilZouitine, 2025-09-21:
Hey could you confirm if this is fixed?😄

Skylerwiernik, 2025-09-21:
Yes. I didn't dig into exactly what fixed it, but the whole modeling pi0 file was completely rewritten and it ran fine with the same config file.