huggingface/lerobot #1992: Could not find 'policy_preprocessor.json' on the HuggingFace Hub at 'lerobot/smolvla_base' when trying to train

cijerezg, 2025-09-21:
### System Info

```Shell
- `lerobot` version: 0.3.4
- Platform: Linux-6.14.0-29-generic-x86_64-with-glibc2.39
- Python version: 3.12.3
- Huggingface_hub version: 0.35.0
- Dataset version: 3.6.0
- Numpy version: 2.2.6
- PyTorch version (GPU?): 2.7.1+cu126 (True)
- Cuda version: 12060
- Using GPU in script?: Yes
```

### Information

- [x] One of the scripts in the examples/ folder of LeRobot
- [ ] My own task or dataset (give details below)

### Reproduction

This is the script I am trying to run:

`
python -m lerobot.scripts.train --policy.path=lerobot/smolvla_base     --dataset.repo_id=cijerezg/pickp-up-v1     --batch_size=128     --steps=20000     --output_dir=outputs/train/my_smolvla_pickup_v3     --job_name=my_smolvla_training     --policy.device=cuda     --wandb.enable=true     --policy.repo_id=pickup_policy_v6     --save_freq=100
`

Not sure if it matters, but I installed everything using uv pip install instead of pip``

### Expected behavior

It should train the smolvla with the dataset, but I get this error:

Entry Not Found for url: https://huggingface.co/lerobot/smolvla_base/resolve/main/policy_preprocessor.json.

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "<frozen runpy>", line 198, in _run_module_as_main
  File "<frozen runpy>", line 88, in _run_code
  File "/home/user/Documents/Research/RL/LeRobot/lerobot/src/lerobot/scripts/train.py", line 346, in <module>
    main()
  File "/home/user/Documents/Research/RL/LeRobot/lerobot/src/lerobot/scripts/train.py", line 342, in main
    train()
  File "/home/user/Documents/Research/RL/LeRobot/lerobot/src/lerobot/configs/parser.py", line 225, in wrapper_inner
    response = fn(cfg, *args, **kwargs)
               ^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/user/Documents/Research/RL/LeRobot/lerobot/src/lerobot/scripts/train.py", line 186, in train
    preprocessor, postprocessor = make_pre_post_processors(
                                  ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/user/Documents/Research/RL/LeRobot/lerobot/src/lerobot/policies/factory.py", line 203, in make_pre_post_processors
    PolicyProcessorPipeline.from_pretrained(
  File "/home/user/Documents/Research/RL/LeRobot/lerobot/src/lerobot/processor/pipeline.py", line 567, in from_pretrained
    loaded_config, base_path = cls._load_config(model_id, config_filename, hub_download_kwargs)
                               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/user/Documents/Research/RL/LeRobot/lerobot/src/lerobot/processor/pipeline.py", line 665, in _load_config
    raise FileNotFoundError(
FileNotFoundError: Could not find 'policy_preprocessor.json' on the HuggingFace Hub at 'lerobot/smolvla_base'``

Effectively, that file doesn't exist, but unsure why it's searching it and how to fix the issue.

AdilZouitine, 2025-09-25:
Hey @cijerezg 👋
Did it solve your issue?

AdilZouitine, 2025-09-26:
Perfect thanks :smile:

cijerezg, 2025-09-27:
@Kexueyi Go to the link shared by @AdilZouitine and download the files, then create a new model in huggingface with those files and that's it. I had to look around where to download the files exactly. You can use my huggingface repo if you want https://huggingface.co/cijerezg/smolvla-test and then you don't call lerobot/smolvla-base but your own repo.

Kexueyi, 2025-10-01:
> [@Kexueyi](https://github.com/Kexueyi) Go to the link shared by [@AdilZouitine](https://github.com/AdilZouitine) and download the files, then create a new model in huggingface with those files and [...]
@cijerezg Thank you so much for providing the instructions! Very helpful!