huggingface/lerobot #2013: terminate called after throwing an instance of 'std::bad_alloc'

yfynb1111, 2025-09-24:
### System Info

```Shell
- `lerobot` version: 0.3.3
- Platform: Linux-6.8.0-83-generic-x86_64-with-glibc2.35
- Python version: 3.11.13
- Huggingface_hub version: 0.35.0
- Dataset version: 3.6.0
- Numpy version: 2.2.6
- PyTorch version (GPU?): 2.10.0.dev20250918+cu128 (True)
- Cuda version: 12080
- Using GPU in script?: <fill in>
```

### Information

- [x] One of the scripts in the examples/ folder of LeRobot
- [ ] My own task or dataset (give details below)

### Reproduction

when run 
dataset = LeRobotDataset(repo_id=args.repo_id, root=args.root)
sample = dataset[0]
it will crash，error log 
terminate called after throwing an instance of 'std::bad_alloc'
  what():  std::bad_alloc

### Expected behavior

hope not to crash

closed 2025-09-24 by yfynb1111

nhnifong, 2025-10-18:
Can you explain what you did to fix it?

cijerezg, 2025-10-18:
@nhnifong 
Install a version of Pytorch that's compatible with cuda. In my case, I installed Pytorch 2.8.0 with cuda 12.8 and locally I have cuda 13.0