huggingface/lerobot #2035: using   "lerobot/svla_so100_stacking"   "lerobot/svla_so100_sorting"  LeRobotDataset    error

axia75, 2025-09-25:
### System Info

```Shell
using   "lerobot/svla_so100_stacking"   "lerobot/svla_so100_sorting"  LeRobotDataset    error

pyarrow.lib.ArrowInvalid: Could not open Parquet input source '<Buffer>': Parquet magic bytes not found in footer. Either the file is corrupted or this is not a parquet file

using lerobot/svla_so101_pickplace 
import torch
from lerobot.policies.smolvla.modeling_smolvla import SmolVLAPolicy
from lerobot.datasets.lerobot_dataset import LeRobotDataset
# policy = SmolVLAPolicy.from_pretrained("E:\models\smolvla_base")
device = "cuda"

dataset = LeRobotDataset("lerobot/svla_so101_pickplace")

# policy = SmolVLAPolicy.from_pretrained("lerobot/smolvla_base")
policy = SmolVLAPolicy.from_pretrained("/home/aigc/.cache/modelscope/hub/lerobot/smolvla_base")
 

observation_state = dataset[0]['observation.state'].unsqueeze(0).to(device)

observation_image_top = dataset[0]['observation.images.top'].unsqueeze(0).to(device)
observation_image_wrist = dataset[0]['observation.images.wrist'].unsqueeze(0).to(device)
# observation_image_top = dataset[0]['observation.images.up'].unsqueeze(0).to(device)
# observation_image_wrist = dataset[0]['observation.images.side'].unsqueeze(0).to(device)

task = dataset[0]['task']

observation = {
    "observation.state": observation_state,
    "observation.image": observation_image_top,
    "observation.image2": observation_image_wrist,
    "task": [task],
}

action = policy.select_action(observation)
print(action)

error:     File "/home/aigc/work/project/cogx/lerobot-main/outputs/eval/diffusion_pusht/test3223.py", line 29, in <module>
    action = policy.select_action(observation)
  File "/install/anaconda3/envs/lerobot/lib/python3.10/site-packages/torch/utils/_contextlib.py", line 116, in decorate_context
    return func(*args, **kwargs)
  File "/home/aigc/work/project/cogx/lerobot-main/src/lerobot/policies/smolvla/modeling_smolvla.py", line 305, in select_action
    actions = self._get_action_chunk(batch, noise)
  File "/home/aigc/work/project/cogx/lerobot-main/src/lerobot/policies/smolvla/modeling_smolvla.py", line 260, in _get_action_chunk
    lang_tokens = batch[f"{OBS_LANGUAGE_TOKENS}"]
KeyError: 'observation.language.tokens'
```

### Information

- [ ] One of the scripts in the examples/ folder of LeRobot
- [ ] My own task or dataset (give details below)

### Reproduction

run code report key Error

### Expected behavior

run

AdilZouitine, 2025-09-25:
We refactored policies to use processors before tokenization happens inside the model. Now it happens in the processor part. :smile:
1. Convert your old model to the new format with this script: [src/lerobot/processor](https://github.com/huggingface/lerobot/blob/main/src/lerobot/processor/migrate_policy_normalization.py).
2. Add a preprocessor before `select_action` and a postprocessor after `select_action`.

Doc: 
1. backward compatibility: https://huggingface.co/docs/lerobot/backwardcomp#policy-normalization-migration-pr-1452
2. Introduction to processor: https://huggingface.co/docs/lerobot/introduction_processors

AdilZouitine, 2025-09-26:
Hey, could you format your message above please? it's hard to read, you have to use backquote 😄

axia75, 2025-09-28:
i   format my message above, please help me