huggingface/lerobot #2061: Compile policies with torch.compile

AdilZouitine, 2025-09-26:
### **Description**
LeRobot currently supports several robotics policies (ACT, Diffusion, PI0, PI0Fast, SmolVLA, TDMPC, VQBet, SAC), but they are not optimized for `torch.compile`.
Adding `torch.compile` support would significantly improve inference and training performance by compiling computational graphs ahead of time.

**Goal**: Make all policies fully compatible with `torch.compile` without graph breaks, demonstrating measurable performance improvements.

### **What Contributors Need to Do**

1. **Run the baseline benchmark** on main branch to establish performance baseline
2. **Implement torch.compile compatibility** following the guidelines below
3. **Submit performance report** showing before/after improvements
4. **Create PR** with your changes and link it to this issue

#### **1. Run Baseline Benchmark**
The script is [here](https://gist.github.com/AdilZouitine/3574664e4cf71605986b49e9148d29ab)
```bash
# Save the benchmark script as benchmark_inference_compile_lerobot.py 
python benchmark_inference_compile_lerobot.py --policy act --device cuda --output baseline_act_report.md
```
This will likely fail with compilation errors - that's expected!

#### **2. Identify and Fix Graph Breaks**
Enable verbose debugging to see what's causing graph breaks:
```python
import torch._dynamo
torch._dynamo.config.verbose = True
torch._dynamo.config.suppress_errors = True  # Don't crash on graph breaks
```

Common issues to fix:
- **Data-dependent control flow**: Replace `if tensor.item() > 0:` with `torch.where(tensor > 0, ...)`
- **Dynamic shapes**: Ensure tensor shapes are consistent across batches
- **Python loops on tensors**: Replace with vectorized operations
- **`.item()` calls**: Keep computations in tensor form
- **In-place operations**: Be careful with operations that modify tensors in-place

#### **3. Test Your Changes**
```bash
python benchmark_inference_compile_lerobot.py --policy act --device cuda --output final_act_report.md
```

#### **4. Create Your PR**
Include:
- Your changes to make the policy compile-compatible
- Both baseline and final benchmark reports
- Brief description of what you fixed

fracapuano, 2025-10-15:
> cases like diffusion/CFM solvers that use long, data-dependent iterative loops

Thank you for your answer! This is very interesting indeed. Are you suggesting SmolVLA inference does not use diffusion/flow-matching like loops? Asking because of how `sample_actions` is implemented, like in here👇 

https://github.com/huggingface/lerobot/blob/a6ff3cfebb0304f2c378515dd30ea06fff8f473f/src/lerobot/policies/smolvla/modeling_smolvla.py#L736-L747

tc-huang, 2025-10-18:
Hello @AdilZouitine,

I've noticed the `pi0fast` policy was deprecated in PR #2203, but it's still included in the benchmark script ([benchmark_inference_compile_lerobot.py](https://gist.github.com/AdilZouitine/3574664e4cf71605986b49e9148d29ab#file-benchmark_inference_compile_lerobot-py)) and this issue's description.

not-heavychevy, 2025-11-15:
I chose SAC as the first policy to implement because of its relatively small size :). I was able to achieve a 2x speedup in training, but it did not improve inference performance.

tommyly201, 2026-01-29:
hi @tc-huang do you want to keep working on ACT so I can do Diffusion instead? Or do you think I can address both?

tommyly201, 2026-01-30:
ok I'll work on Diffusion instead and
will help however I can on your PR

kostasrigatos, 2026-09-25:
Hi everyone, I've worked on TD-MPC for this issue: #4776 makes the MPC planner compile as a single graph with no graph breaks and no recompiles (refactored warm-start state out of the compiled path, and moved sampling indices onto the model's device so CUDA-graph modes work). Compiled planning is ~1.2× faster than eager on a T4; benchmark and tests are in the PR. It builds on a small TD-MPC bug fix, #4741.